gbionics/robot-log-visualizer #29: Unable to install it in WSL2 Ubuntu 20.04

Nicogene, 2022-04-04:
I have followed the README instructions but I am not able to install it

```
$ pip install robot-log-visualizer
ERROR: Could not find a version that satisfies the requirement robot-log-visualizer (from versions: none)
ERROR: No matching distribution found for robot-log-visualizer
```

same thing for:

```
python3 -m pip install git+https://github.com/ami-iit/robot-log-visualizer.git
Collecting git+https://github.com/ami-iit/robot-log-visualizer.git
  Cloning https://github.com/ami-iit/robot-log-visualizer.git to /tmp/pip-req-build-8w6eby2v
  Running command git clone -q https://github.com/ami-iit/robot-log-visualizer.git /tmp/pip-req-build-8w6eby2v
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /usr/share/python-wheels/pep517-0.8.2-py2.py3-none-any.whl/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpgcqv6gkt
       cwd: /tmp/pip-req-build-8w6eby2v
  Complete output (1 lines):
  /usr/bin/python3: can't find '__main__' module in '/usr/share/python-wheels/pep517-0.8.2-py2.py3-none-any.whl/pep517/_in_process.py'
  ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 /usr/share/python-wheels/pep517-0.8.2-py2.py3-none-any.whl/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpgcqv6gkt Check the logs for full command output.
```

Nicogene, 2022-04-04:
I discovered that `python 3.8` is required

https://github.com/ami-iit/robot-log-visualizer/blob/main/setup.cfg#L41

I can open a PR for adding it in the readme.

I had to change the default python to 3.8 changing the user alternatives (it was 3.6).
Outside the `virtualenv` it does not show anymore the error but it get stuck

Inside the virtual env I get this error:

```
Collecting robot-log-visualizer
  Downloading robot_log_visualizer-0.1.0-py3-none-any.whl (103 kB)
     |████████████████████████████████| 103 kB 3.9 MB/s
Requirement already satisfied: idyntree>=5.0.1 in ./robotology/robotology-superbuild/build/install/lib/python3/dist-packages (from robot-log-visualizer) (5.1.0)
Collecting pyqtconsole
  Downloading pyqtconsole-1.2.2-py2.py3-none-any.whl (20 kB)
Collecting matplotlib
  Downloading matplotlib-3.5.1-cp38-cp38-manylinux_2_5_x86_64.manylinux1_x86_64.whl (11.3 MB)
     |████████████████████████████████| 11.3 MB 13.9 MB/s
Collecting PyQt5
  Downloading PyQt5-5.15.6-cp36-abi3-manylinux1_x86_64.whl (8.3 MB)
     |████████████████████████████████| 8.3 MB 12.1 MB/s
Collecting h5py
  Downloading h5py-3.6.0-cp38-cp38-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (4.5 MB)
     |████████████████████████████████| 4.5 MB 15.3 MB/s
Requirement already satisfied: icub-models in ./robotology/robotology-superbuild/build/install/lib/python3/dist-packages (from robot-log-visualizer) (1.23.0)
Collecting numpy
  Downloading numpy-1.22.3-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (16.8 MB)
     |████████████████████████████████| 16.8 MB 553 kB/s
Collecting PyQtWebEngine
  Downloading PyQtWebEngine-5.15.5-cp36-abi3-manylinux1_x86_64.whl (228 kB)
     |████████████████████████████████| 228 kB 5.1 MB/s
Collecting meshcat
  Downloading meshcat-0.3.2-py3-none-any.whl (2.6 MB)
     |████████████████████████████████| 2.6 MB 4.4 MB/s
Collecting qtpy
  Downloading QtPy-2.0.1-py3-none-any.whl (65 kB)
     |████████████████████████████████| 65 kB 2.4 MB/s
Collecting jedi
  Downloading jedi-0.18.1-py2.py3-none-any.whl (1.6 MB)
     |████████████████████████████████| 1.6 MB 10.1 MB/s
Collecting python-dateutil>=2.7
  Using cached python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
Collecting kiwisolver>=1.0.1
  Downloading kiwisolver-1.4.2-cp38-cp38-manylinux_2_5_x86_64.manylinux1_x86_64.whl (1.2 MB)
     |████████████████████████████████| 1.2 MB 9.4 MB/s
Collecting pillow>=6.2.0
  Downloading Pillow-9.1.0-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (4.3 MB)
     |████████████████████████████████| 4.3 MB 9.7 MB/s
Collecting fonttools>=4.22.0
  Downloading fonttools-4.31.2-py3-none-any.whl (899 kB)
     |████████████████████████████████| 899 kB 14.2 MB/s
Collecting packaging>=20.0
  Using cached packaging-21.3-py3-none-any.whl (40 kB)
Collecting cycler>=0.10
  Downloading cycler-0.11.0-py3-none-any.whl (6.4 kB)
Collecting pyparsing>=2.2.1
  Using cached pyparsing-3.0.7-py3-none-any.whl (98 kB)
Collecting PyQt5-Qt5>=5.15.2
  Downloading PyQt5_Qt5-5.15.2-py3-none-manylinux2014_x86_64.whl (59.9 MB)
     |████████████████████████████████| 59.9 MB 276 kB/s
Collecting PyQt5-sip<13,>=12.8
  Downloading PyQt5_sip-12.9.1-cp38-cp38-manylinux1_x86_64.whl (353 kB)
     |████████████████████████████████| 353 kB 4.5 MB/s
Collecting PyQtWebEngine-Qt5>=5.15.2
  Downloading PyQtWebEngine_Qt5-5.15.2-py3-none-manylinux2014_x86_64.whl (67.5 MB)
     |████████████████████████████████| 67.5 MB 7.4 MB/s
Collecting pyngrok>=4.1.6
[... 49 lines cut ...]
  Downloading asttokens-2.0.5-py2.py3-none-any.whl (20 kB)
Collecting pure-eval
  Downloading pure_eval-0.2.2-py3-none-any.whl (11 kB)
Collecting wcwidth
  Downloading wcwidth-0.2.5-py2.py3-none-any.whl (30 kB)
Building wheels for collected packages: pyngrok
  Building wheel for pyngrok (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/ngenesio/visualizer-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-epru3vh1/pyngrok/setup.py'"'"'; __file__='"'"'/tmp/pip-install-epru3vh1/pyngrok/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-9h2a_q6m
       cwd: /tmp/pip-install-epru3vh1/pyngrok/
  Complete output (6 lines):
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help

  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for pyngrok
  Running setup.py clean for pyngrok
Failed to build pyngrok
Installing collected packages: pyparsing, packaging, qtpy, parso, jedi, pyqtconsole, six, python-dateutil, kiwisolver, pillow, fonttools, cycler, numpy, matplotlib, PyQt5-Qt5, PyQt5-sip, PyQt5, h5py, PyQtWebEngine-Qt5, PyQtWebEngine, PyYAML, pyngrok, pyzmq, ptyprocess, pexpect, traitlets, executing, asttokens, pure-eval, stack-data, decorator, wcwidth, prompt-toolkit, backcall, matplotlib-inline, pygments, pickleshare, ipython, u-msgpack-python, tornado, meshcat, robot-log-visualizer
    Running setup.py install for pyngrok ... done
Successfully installed PyQt5-5.15.6 PyQt5-Qt5-5.15.2 PyQt5-sip-12.9.1 PyQtWebEngine-5.15.5 PyQtWebEngine-Qt5-5.15.2 PyYAML-6.0 asttokens-2.0.5 backcall-0.2.0 cycler-0.11.0 decorator-5.1.1 executing-0.8.3 fonttools-4.31.2 h5py-3.6.0 ipython-8.2.0 jedi-0.18.1 kiwisolver-1.4.2 matplotlib-3.5.1 matplotlib-inline-0.1.3 meshcat-0.3.2 numpy-1.22.3 packaging-21.3 parso-0.8.3 pexpect-4.8.0 pickleshare-0.7.5 pillow-9.1.0 prompt-toolkit-3.0.29 ptyprocess-0.7.0 pure-eval-0.2.2 pygments-2.11.2 pyngrok-5.1.0 pyparsing-3.0.7 pyqtconsole-1.2.2 python-dateutil-2.8.2 pyzmq-22.3.0 qtpy-2.0.1 robot-log-visualizer-0.1.0 six-1.16.0 stack-data-0.2.0 tornado-6.1 traitlets-5.1.1 u-msgpack-python-2.7.1 wcwidth-0.2.5
```

traversaro, 2022-04-04:
> I had to change the default python to 3.8 changing the user alternatives (it was 3.6).

Interesting, do you have any idea why that happened? The default one should be 3.8 (see https://askubuntu.com/questions/1232812/whats-the-default-python-version-in-ubuntu-20-04). Perhaps this was a system that was updated from 18.04 ?

Nicogene, 2022-04-05:
> Interesting, do you have any idea why that happened? The default one should be 3.8 (see https://askubuntu.com/questions/1232812/whats-the-default-python-version-in-ubuntu-20-04). Perhaps this was a system that was updated from 18.04 ?

Probably that's the reason why.

I managed to install it on wsl running an X server because apparently pip grabs the display 🤦‍♂️

See https://stackoverflow.com/questions/33925566/pip-install-hangs

At the end the issue was a mix of problems of my setup and pip strange behaviour, closing.

closed 2022-04-05 by Nicogene