gbionics/robot-log-visualizer #36: Unable to load a model that contains geometric primitives

Nicogene, 2022-04-07:
It seems that `robot-log-visualizer` supports only external meshes, a model that contains geometric primitives gives this error:

```
  File "/home/ngenesio/.local/lib/python3.8/site-packages/robot_log_visualizer/ui/gui.py", line 374, in open_mat_file
    self.meshcat_provider.load_model(
  File "/home/ngenesio/.local/lib/python3.8/site-packages/robot_log_visualizer/robot_visualizer/meshcat_provider.py", line 42, in load_model
    self.meshcat_visualizer.load_model_from_file(
  File "/home/ngenesio/.local/lib/python3.8/site-packages/robot_log_visualizer/robot_visualizer/meshcat_visualizer.py", line 379, in load_model_from_file
    self.load_model(model=model_loader.model(), model_name=model_name, color=color)
  File "/home/ngenesio/.local/lib/python3.8/site-packages/robot_log_visualizer/robot_visualizer/meshcat_visualizer.py", line 396, in load_model
    self.__add_model_geometry_to_viewer(
  File "/home/ngenesio/.local/lib/python3.8/site-packages/robot_log_visualizer/robot_visualizer/meshcat_visualizer.py", line 198, in __add_model_geometry_to_viewer
    + solid_shape.asExternalMesh().getName()
```

traversaro, 2022-04-07:
Probably there was a fix for that in the copy of that code in idyntree?

GiulioRomualdi, 2022-04-07:
This logic was not implemented. I developed in https://github.com/ami-iit/robot-log-visualizer/tree/support_robot_without_meshses

GiulioRomualdi, 2022-04-07:
Fixed with 14b5e32ed51fcb6b03d032c89a8ec18124d5e4f7 , 3a478a61b9aafc794abbb5959cb2643a08c6c9a5 and https://github.com/ami-iit/robot-log-visualizer/commit/e6d4ecc8ebfb6d66532ff6070ac31f066c5af185

closed 2022-04-07 by GiulioRomualdi

traversaro, 2022-04-07:
> > Probably there was a fix for that in the copy of that code in idyntree?
> 
> This problem affects also the idyntree code

Ack, probably I was confusing this with the similar issue for irrlicht-based virtualizer.