gbionics/robot-log-visualizer #43: Icons are not correctly loaded if the dependencies are installed from apt

GiulioRomualdi, 2022-05-14:
If I installed the visualizer from pip the incons are correctly loaded
```
pip list
Package              Version
-------------------- -------
asttokens            2.0.5
backcall             0.2.0
cycler               0.11.0
decorator            5.1.1
executing            0.8.3
fonttools            4.33.3
h5py                 3.6.0
icub-models          1.23.4
idyntree             5.2.0
ipython              8.3.0
jedi                 0.18.1
kiwisolver           1.4.2
matplotlib           3.5.2
matplotlib-inline    0.1.3
meshcat              0.3.2
numpy                1.22.3
packaging            21.3
parso                0.8.3
pexpect              4.8.0
pickleshare          0.7.5
Pillow               9.1.0
pip                  22.0.2
prompt-toolkit       3.0.29
ptyprocess           0.7.0
pure-eval            0.2.2
Pygments             2.12.0
pyngrok              5.1.0
pyparsing            3.0.9
PyQt5                5.15.6
PyQt5-Qt5            5.15.2
PyQt5-sip            12.10.1
pyqtconsole          1.2.2
PyQtWebEngine        5.15.5
PyQtWebEngine-Qt5    5.15.2
python-dateutil      2.8.2
PyYAML               6.0
pyzmq                22.3.0
QtPy                 2.1.0
robot-log-visualizer 0.1.5
setuptools           59.6.0
six                  1.16.0
stack-data           0.2.0
tornado              6.1
traitlets            5.2.0
u-msgpack-python     2.7.1
wcwidth              0.2.5
```

On the other hand If I installed from the superbuild I got the following error
```
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
```
and some of the icons are not shown correctly. In this context pyqt5 is installed from apt. I'm experecing the issue on ubuntu 22.04. Notice that in ubuntu 22.04, the version of pyqt5 is 5.15.6 (the same of pip) (https://repology.org/project/python:pyqt5/versions)

Releavant issue: https://github.com/robotology/robotology-superbuild/issues/1118

traversaro, 2022-05-15:
Related issue: https://github.com/ami-iit/robot-log-visualizer/issues/40 . 
Can you report the output of `pip list --verbose` ? It can be helpful to understand which dependencies where installed by apt and which one via pip.

GiulioRomualdi, 2023-03-10:
Closed thanks to #42

closed 2023-03-10 by GiulioRomualdi