gbionics/robot-log-visualizer #46: Unable to run the visualizer on windows when it is installed from conda

GiulioRomualdi, 2022-09-07:
I installed robot log visualizer by running
```console
mamba install -c robotology robot-log-visualizer
```
when I tried to run the application I got the following error
```
Traceback (most recent call last):
  File "C:\Users\gromualdi\AppData\Local\mambaforge\envs\test-visualizer\Scripts\robot-log-visualizer-script.py", line 6, in <module>
    from robot_log_visualizer.__main__ import main
  File "C:\Users\gromualdi\AppData\Local\mambaforge\envs\test-visualizer\lib\site-packages\robot_log_visualizer\__main__.py", line 14, in <module>
    from robot_log_visualizer.file_reader.signal_provider import SignalProvider
  File "C:\Users\gromualdi\AppData\Local\mambaforge\envs\test-visualizer\lib\site-packages\robot_log_visualizer\file_reader\signal_provider.py", line 7, in <module>
    import h5py
  File "C:\Users\gromualdi\AppData\Local\mambaforge\envs\test-visualizer\lib\site-packages\h5py\__init__.py", line 33, in <module>
    from . import version
  File "C:\Users\gromualdi\AppData\Local\mambaforge\envs\test-visualizer\lib\site-packages\h5py\version.py", line 15, in <module>
    from . import h5 as _h5
  File "h5py\h5.pyx", line 1, in init h5py.h5
ImportError: DLL load failed while importing defs: The specified procedure could not be found.
```

```
name: test-visualizer
channels:
  - robotology
  - conda-forge
dependencies:
  - assimp=5.2.4=hc2aa0de_0
  - asttokens=2.0.8=pyhd8ed1ab_0
  - backcall=0.2.0=pyh9f0ad1d_0
  - backports=1.0=py_2
  - backports.functools_lru_cache=1.6.4=pyhd8ed1ab_0
  - boost-cpp=1.74.0=h9f4b32c_8
  - brotli=1.0.9=h8ffe710_7
  - brotli-bin=1.0.9=h8ffe710_7
  - bzip2=1.0.8=h8ffe710_4
  - ca-certificates=2022.6.15=h5b45459_0
  - cached-property=1.5.2=hd8ed1ab_1
  - cached_property=1.5.2=pyha770c72_1
  - certifi=2022.6.15=pyhd8ed1ab_1
  - colorama=0.4.5=pyhd8ed1ab_0
  - cycler=0.11.0=pyhd8ed1ab_0
  - decorator=5.1.1=pyhd8ed1ab_0
  - eigen=3.4.0=h2d74725_0
  - executing=1.0.0=pyhd8ed1ab_0
  - fonttools=4.37.1=py310he2412df_0
  - freetype=2.12.1=h546665d_0
  - gettext=0.19.8.1=ha2e2712_1008
  - glib=2.72.1=h7755175_0
  - glib-tools=2.72.1=h7755175_0
  - gst-plugins-base=1.20.3=h001b923_1
  - gstreamer=1.20.3=h6b5321d_1
  - h5py=3.7.0=nompi_py310h425bae0_101
  - hdf5=1.12.2=nompi_h2a0e4a3_100
  - icu=70.1=h0e60522_0
  - icub-models=1.25.0=py310h8a704f9_0
  - idyntree=6.1.0=py310h2bb8f68_1
  - intel-openmp=2022.1.0=h57928b3_3787
  - ipopt=3.14.8=h99cadeb_0
  - ipython=8.5.0=pyh08f2357_1
  - irrlicht=1.8.5=h739eaf8_2
  - jedi=0.18.1=pyhd8ed1ab_2
  - jpeg=9e=h8ffe710_2
  - kiwisolver=1.4.4=py310h476a331_0
  - krb5=1.19.3=h1176d77_0
  - lcms2=2.12=h2a16943_0
  - lerc=4.0.0=h63175ca_0
  - libblas=3.9.0=16_win64_mkl
  - libbrotlicommon=1.0.9=h8ffe710_7
  - libbrotlidec=1.0.9=h8ffe710_7
  - libbrotlienc=1.0.9=h8ffe710_7
  - libcblas=3.9.0=16_win64_mkl
  - libclang=14.0.6=default_h77d9078_0
  - libclang13=14.0.6=default_h77d9078_0
  - libcurl=7.83.1=h789b8ee_0
  - libdeflate=1.13=h8ffe710_0
  - libffi=3.4.2=h8ffe710_5
  - libflang=5.0.0=h6538335_20180525
  - libglib=2.72.1=h3be07f2_0
  - libiconv=1.16=he774522_0
  - liblapack=3.9.0=16_win64_mkl
  - libogg=1.3.4=h8ffe710_1
  - libosqp=0.6.2=h0e60522_3
  - libpng=1.6.37=h1d00b33_4
  - libsodium=1.0.18=h8d14728_1
  - libsqlite=3.39.3=hcfcfb64_0
  - libssh2=1.10.0=h680486a_3
  - libtiff=4.4.0=h92677e6_3
  - libvorbis=1.3.7=h0e60522_0
  - libwebp=1.2.4=h8ffe710_0
  - libwebp-base=1.2.4=h8ffe710_0
  - libxcb=1.13=hcd874cb_1004
  - libxml2=2.9.14=hf5bbc77_4
  - libzlib=1.2.12=h8ffe710_2
  - llvm-meta=5.0.0=0
  - m2w64-gcc-libgfortran=5.3.0=6
  - m2w64-gcc-libs=5.3.0=7
  - m2w64-gcc-libs-core=5.3.0=7
  - m2w64-gmp=6.1.0=2
  - m2w64-libwinpthread-git=5.0.0.4634.697f757=2
  - matplotlib=3.5.3=py310h5588dad_2
  - matplotlib-base=3.5.3=py310h7329aa0_2
  - matplotlib-inline=0.1.6=pyhd8ed1ab_0
  - meshcat-python=0.3.2=pyhd8ed1ab_0
  - metis=5.1.0=h0e60522_1006
  - mkl=2022.1.0=h6a75c08_874
  - msys2-conda-epoch=20160418=1
  - mumps-seq=5.2.1=hb3f9cae_11
  - munkres=1.1.4=pyh9f0ad1d_0
  - numpy=1.23.2=py310h8a5b91a_0
  - openjpeg=2.5.0=hc9384bd_1
  - openmp=5.0.0=vc14_1
  - openssl=1.1.1q=h8ffe710_0
  - osqp-eigen=0.7.0=h0e60522_0
  - packaging=21.3=pyhd8ed1ab_0
  - parso=0.8.3=pyhd8ed1ab_0
  - pcre=8.45=h0e60522_0
  - pickleshare=0.7.5=py_1003
  - pillow=9.2.0=py310h52929f7_2
  - pip=22.2.2=pyhd8ed1ab_0
  - ply=3.11=py_1
  - prompt-toolkit=3.0.31=pyha770c72_0
  - pthread-stubs=0.4=hcd874cb_1001
  - pure_eval=0.2.2=pyhd8ed1ab_0
  - pygments=2.13.0=pyhd8ed1ab_0
  - pyngrok=5.1.0=pyhd8ed1ab_0
  - pyparsing=3.0.9=pyhd8ed1ab_0
  - pyqt=5.15.7=py310hbabf5d4_0
  - pyqt5-sip=12.11.0=py310h8a704f9_0
  - pyqtconsole=1.2.2=pyhd8ed1ab_0
  - pyqtwebengine=5.15.7=py310hbabf5d4_0
  - python=3.10.6=h9a09f29_0_cpython
  - python-dateutil=2.8.2=pyhd8ed1ab_0
  - python_abi=3.10=2_cp310
  - pyyaml=6.0=py310he2412df_4
  - pyzmq=23.2.1=py310h73ada01_0
  - qt-main=5.15.4=h467ea89_2
  - qt-webengine=5.15.4=h7325c83_3
  - qtpy=2.2.0=pyhd8ed1ab_0
  - robot-log-visualizer=0.2.2=py310hbbfc1a7_64
  - sdl=1.2.52=h0e60522_1
  - sdl2=2.0.22=h0e60522_2
  - setuptools=65.3.0=pyhd8ed1ab_1
  - sip=6.6.2=py310h8a704f9_0
  - six=1.16.0=pyh6c4a22f_0
  - sqlite=3.39.3=hcfcfb64_0
  - stack_data=0.5.0=pyhd8ed1ab_0
  - tbb=2021.5.0=h91493d7_2
  - tk=8.6.12=h8ffe710_0
  - toml=0.10.2=pyhd8ed1ab_0
  - tornado=6.2=py310he2412df_0
  - traitlets=5.3.0=pyhd8ed1ab_0
  - tzdata=2022c=h191b570_0
  - u-msgpack-python=2.7.1=pyh9f0ad1d_0
  - ucrt=10.0.20348.0=h57928b3_0
  - unicodedata2=14.0.0=py310he2412df_1
  - vc=14.2=hb210afc_7
  - vs2015_runtime=14.29.30139=h890b9b1_7
  - wcwidth=0.2.5=pyh9f0ad1d_2
  - wheel=0.37.1=pyhd8ed1ab_0
  - xorg-libxau=1.0.9=hcd874cb_0
  - xorg-libxdmcp=1.1.3=hcd874cb_0
  - xz=5.2.6=h8d14728_0
  - yaml=0.2.5=h8ffe710_2
  - zeromq=4.3.4=h0e60522_1
  - zlib=1.2.12=h8ffe710_2
  - zstd=1.5.2=h7755175_4
prefix: C:\Users\gromualdi\AppData\Local\mambaforge\envs\test-visualizer
```

cc @traversaro

GiulioRomualdi, 2022-09-07:
Accordingly to the error there seems to be something wrong with h5py however I'm able to import it in python whith 
```python
import h5py
```

GiulioRomualdi, 2022-09-07:
Possibly related to https://stackoverflow.com/questions/68296054/importerror-dll-load-failed-while-importing-defs-the-specified-procedure-could

GiulioRomualdi, 2022-09-07:
cc @FrancescaBruzzone

traversaro, 2022-09-07:
I have exactly the same environment, and `robot-log-visualizer` works fine for me.

This is the diff between the `conda env export` reported in the issue and mine:
~~~
--- <unnamed>
+++ <unnamed>
@@ -1,4 +1,4 @@
-name: test-visualizer
+name: rlv
 channels:
   - robotology
   - conda-forge
@@ -143,4 +143,4 @@
   - zeromq=4.3.4=h0e60522_1
   - zlib=1.2.12=h8ffe710_2
   - zstd=1.5.2=h7755175_4
-prefix: C:\Users\gromualdi\AppData\Local\mambaforge\envs\test-visualizer
+prefix: C:\Users\STraversaro\AppData\Local\mambaforge\envs\rlv
~~~

antonellopaolino, 2022-11-18:
I had the same problem of @GiulioRomualdi, I solved exporting the `.mat` file using `'-v7.3'` in matlab command window, as found in https://github.com/h5py/h5py/issues/757#issuecomment-823757614.

traversaro, 2022-11-18:
@antonellopaolino can you report how you installed robot-log-visualizer ? It could be something strange on your libmatio installation/build.

antonellopaolino, 2022-11-18:
Same command

> I installed robot log visualizer by running
> 
> ```
> mamba install -c robotology robot-log-visualizer
> ```

traversaro, 2022-11-18:
Can you report the output of mamba list?

GiulioRomualdi, 2022-11-18:
Hi @antonellopaolino, if you saved the mat file with YarpRobotLoggerDevice it should be already compliant with h5py (i.e.,  v7.3)

antonellopaolino, 2022-11-21:
> Can you report the output of mamba list?

```
(testvisualizer) C:\Users\apaolino>mamba list
# packages in environment at C:\Users\apaolino\AppData\Local\mambaforge\envs\testvisualizer:
#
# Name                    Version                   Build  Channel
assimp                    5.2.4                hc2aa0de_0    conda-forge
asttokens                 2.0.5              pyhd3eb1b0_0
backcall                  0.2.0              pyhd3eb1b0_0
boost-cpp                 1.74.0               h9f4b32c_8    conda-forge
brotli                    1.0.9                h2bbff1b_7
brotli-bin                1.0.9                h2bbff1b_7
bzip2                     1.0.8                he774522_0
ca-certificates           2022.10.11           haa95532_0
certifi                   2022.9.24        py38haa95532_0
colorama                  0.4.5            py38haa95532_0
cycler                    0.11.0             pyhd3eb1b0_0
decorator                 5.1.1              pyhd3eb1b0_0
eigen                     3.3.7                h59b6b97_1
executing                 0.8.3              pyhd3eb1b0_0
fonttools                 4.25.0             pyhd3eb1b0_0
freetype                  2.10.4               hd328e21_0
glib                      2.69.1               h5dc1a3c_1
gst-plugins-base          1.18.5               h9e645db_0
gstreamer                 1.18.5               hd78058f_0
h5py                      3.7.0            py38h3de5c98_0
hdf5                      1.10.6               h1756f20_1
icc_rt                    2022.1.0             h6049295_2
icu                       58.2                 ha925a31_3
icub-models               1.25.0           py38h885f38d_0    conda-forge
idyntree                  6.1.0            py38h1bee681_0    conda-forge
intel-openmp              2022.1.0          h59b6b97_3788
ipopt                     3.14.8               h99cadeb_0    conda-forge
ipython                   8.6.0            py38haa95532_0
irrlicht                  1.8.5                h739eaf8_2    conda-forge
jedi                      0.18.1           py38haa95532_1
jpeg                      9e                   h2bbff1b_0
kiwisolver                1.4.2            py38hd77b12b_0
lerc                      3.0                  hd77b12b_0
libblas                   3.9.0              16_win64_mkl    conda-forge
libbrotlicommon           1.0.9                h2bbff1b_7
libbrotlidec              1.0.9                h2bbff1b_7
libbrotlienc              1.0.9                h2bbff1b_7
libcblas                  3.9.0              16_win64_mkl    conda-forge
libclang                  12.0.0          default_h627e005_2
libdeflate                1.8                  h2bbff1b_5
libffi                    3.4.2                hd77b12b_4
libflang                  5.0.0           h6538335_20180525    conda-forge
libiconv                  1.16                 h2bbff1b_2
liblapack                 3.9.0              16_win64_mkl    conda-forge
libogg                    1.3.5                h2bbff1b_1
libosqp                   0.6.2                h0e60522_3    conda-forge
libpng                    1.6.37               h2a8f88b_0
libsodium                 1.0.18               h62dcd97_0
libtiff                   4.4.0                h8a3f274_2
libvorbis                 1.3.7                he774522_0
libwebp                   1.2.4                h2bbff1b_0
libwebp-base              1.2.4                h2bbff1b_0
libxml2                   2.9.14               h0ad7f3c_0
libxslt                   1.1.35               h2bbff1b_0
libzlib                   1.2.12               h8ffe710_2    conda-forge
llvm-meta                 5.0.0                         0    conda-forge
lz4-c                     1.9.3                h2bbff1b_1
matplotlib                3.5.3            py38haa95532_0
matplotlib-base           3.5.3            py38he529843_0    conda-forge
matplotlib-inline         0.1.6            py38haa95532_0
meshcat-python            0.3.2              pyhd8ed1ab_0    conda-forge
metis                     5.1.0                h6538335_4
mkl                       2022.1.0           h6a75c08_874    conda-forge
[... 31 lines cut ...]
robot-log-visualizer      0.2.2           py38h4317176_64    robotology
sdl                       1.2.52               h0e60522_1    conda-forge
sdl2                      2.0.22               h0e60522_2    conda-forge
setuptools                65.5.0           py38haa95532_0
sip                       6.6.2            py38hd77b12b_0
six                       1.16.0             pyhd3eb1b0_1
sqlite                    3.39.3               h2bbff1b_0
stack_data                0.2.0              pyhd3eb1b0_0
tbb                       2021.6.0             h59b6b97_0
tk                        8.6.12               h2bbff1b_0
toml                      0.10.2             pyhd3eb1b0_0
tornado                   6.2              py38h2bbff1b_0
traitlets                 5.1.1              pyhd3eb1b0_0
u-msgpack-python          2.7.2              pyhd8ed1ab_0    conda-forge
vc                        14.2                 h21ff451_1
vs2015_runtime            14.27.29016          h5e58377_2
wcwidth                   0.2.5              pyhd3eb1b0_0
wheel                     0.37.1             pyhd3eb1b0_0
wincertstore              0.2              py38haa95532_2
xz                        5.2.6                h8cc25b3_0
yaml                      0.2.5                he774522_0
zeromq                    4.3.4                hd77b12b_0
zlib                      1.2.12               h8cc25b3_2
zstd                      1.5.2                h19a0ad4_0
```

antonellopaolino, 2022-11-21:
> Hi @antonellopaolino, if you saved the mat file with YarpRobotLoggerDevice it should be already compliant with h5py (i.e., v7.3)

I generated it converting the data format from a Simulink workspace saved during iRonCub experiments, since I was saving it with a default version it was not working before with the same error reported by @GiulioRomualdi, but after specifying the 7.3 version in matlab it worked for me without any problems.