gbionics/robot-log-visualizer #48: Error when running visualizer on Ubuntu 20.04

gabrielenava, 2022-11-23:
I tried to install `robot-log-visualizer` on my Ubuntu 20.04 LTS 

I tried:

https://github.com/ami-iit/robot-log-visualizer#-install-from-pip-with-apt-python

https://github.com/ami-iit/robot-log-visualizer#-use-the-appimage

and 

```
mamba install -c robotology robot-log-visualizer
```

in all cases, I got the following error when running the visualizer:

```python
You can open the visualizer by visiting the following URL:
http://127.0.0.1:7000/static/
Traceback (most recent call last):
  File "/home/gnava/Software/python_projets/MPC/venv/bin/robot-log-visualizer", line 8, in <module>
    sys.exit(main())
  File "/home/gnava/Software/python_projets/MPC/venv/lib/python3.8/site-packages/robot_log_visualizer/__main__.py", line 38, in main
    gui = RobotViewerMainWindow(
  File "/home/gnava/Software/python_projets/MPC/venv/lib/python3.8/site-packages/robot_log_visualizer/ui/gui.py", line 122, in __init__
    self.toolButton_on_click()
  File "/home/gnava/Software/python_projets/MPC/venv/lib/python3.8/site-packages/robot_log_visualizer/ui/gui.py", line 248, in toolButton_on_click
    PlotItem(signal_provider=self.signal_provider, period=self.animation_period)
  File "/home/gnava/Software/python_projets/MPC/venv/lib/python3.8/site-packages/robot_log_visualizer/ui/plot_item.py", line 17, in __init__
    self.canvas = MatplotlibViewerCanvas(
  File "/home/gnava/Software/python_projets/MPC/venv/lib/python3.8/site-packages/robot_log_visualizer/plotter/matplotlib_viewer_canvas.py", line 45, in __init__
    self.vertical_line_anim = animation.FuncAnimation(
  File "/home/gnava/Software/python_projets/MPC/venv/lib/python3.8/site-packages/matplotlib/animation.py", line 1659, in __init__
    super().__init__(fig, **kwargs)
  File "/home/gnava/Software/python_projets/MPC/venv/lib/python3.8/site-packages/matplotlib/animation.py", line 1394, in __init__
    super().__init__(fig, event_source=event_source, *args, **kwargs)
  File "/home/gnava/Software/python_projets/MPC/venv/lib/python3.8/site-packages/matplotlib/animation.py", line 875, in __init__
    self._setup_blit()
  File "/home/gnava/Software/python_projets/MPC/venv/lib/python3.8/site-packages/matplotlib/animation.py", line 1192, in _setup_blit
    self._init_draw()
  File "/home/gnava/Software/python_projets/MPC/venv/lib/python3.8/site-packages/matplotlib/animation.py", line 1723, in _init_draw
    self._drawn_artists = self._init_func()
  File "/home/gnava/Software/python_projets/MPC/venv/lib/python3.8/site-packages/robot_log_visualizer/plotter/matplotlib_viewer_canvas.py", line 124, in init_vertical_line
    return self.vertical_line, *(self.active_paths.values())
AttributeError: 'MatplotlibViewerCanvas' object has no attribute 'active_paths'

```

which is strange, as the attribute `active_paths` is defined inside the class and looks indented correctly.

traversaro, 2022-11-23:
Can you run `apt list --installed`, and `pip list --verbose`  and `mamba list`  in all the environment in which you tried? Thanks

gabrielenava, 2022-11-23:
### Mamba

[apt_list_mamba.txt](https://github.com/ami-iit/robot-log-visualizer/files/10074569/apt_list_mamba.txt)
[pip_list_mamba.txt](https://github.com/ami-iit/robot-log-visualizer/files/10074571/pip_list_mamba.txt)
[mamba_list.txt](https://github.com/ami-iit/robot-log-visualizer/files/10074570/mamba_list.txt)

## Python project environment

[apt_list_python.txt](https://github.com/ami-iit/robot-log-visualizer/files/10074627/apt_list_python.txt)
[pip_list_python.txt](https://github.com/ami-iit/robot-log-visualizer/files/10074628/pip_list_python.txt)

and there in no mamba list in this case

HosameldinMohamed, 2022-11-25:
I confirm having the same issue when trying to run the visualizer.

I'm using the installation via `robotology-superbuild` source compilation and the dependencies are handled by Conda. The OS is Arch Linux. There could be some interference with system libraries but I made sure that `Python` and `PyQt5` packages used are from Conda. I tried both versions `0.2.2` and `0.3.0` and they both produce the same error.

Interestingly. Using the `AppImage` installation, I got the error while installing version `0.3.0`, but **not** with version `0.2.2`, where it seems to be working fine!

traversaro, 2022-11-25:
I am not sure what is happening, but can you try to move the line `self.active_paths = {}` from https://github.com/ami-iit/robot-log-visualizer/blob/82aa5ba1c92599672365b5393d520a049e28a3ac/robot_log_visualizer/plotter/matplotlib_viewer_canvas.py#L54 before the block:
~~~
        self.vertical_line_anim = animation.FuncAnimation(
            self.fig,
            self.update_vertical_line,
            init_func=self.init_vertical_line,
            interval=self.periond_in_ms,
            blit=True,
        )
~~~

?

HosameldinMohamed, 2022-11-27:
> I am not sure what is happening, but can you try to move the line `self.active_paths = {}` 

I moved the line and re-installed the app, and it works with no error..

I can't see the robot's visualizer though, I can only see it from the browser if I use the provided URL. I don't think it's related to this modification though.

gabrielenava, 2022-11-30:
[this suggestion](https://github.com/ami-iit/robot-log-visualizer/issues/48#issuecomment-1327726236) worked for me, now I can open the visualizer!

[image]

traversaro, 2022-11-30:
Can anyone open the PR with the fix? Thanks!

gabrielenava, 2022-12-01:
PR opened: https://github.com/ami-iit/robot-log-visualizer/pull/49

closed 2022-12-14 by GiulioRomualdi