gbionics/robot-log-visualizer #57: Crash if the dimension of the data in the "log" is 1x1

davidegorbani, 2023-07-14:
The `robot-log-visualizer` crashes if in the `log` field of the matlab structure generated by the `YarpRobotLogger` there is a data field of dimension 1x1 such as the one shown in the picture.

[image]

The error that I get is the following:

```
Traceback (most recent call last):
  File "/home/dgorbani/software/robotology-superbuild/build/install/lib/python3/dist-packages/robot_log_visualizer/ui/gui.py", line 627, in open_mat_file
    self.__load_mat_file(file_name)
  File "/home/dgorbani/software/robotology-superbuild/build/install/lib/python3/dist-packages/robot_log_visualizer/ui/gui.py", line 547, in __load_mat_file
    self.signal_provider.open_mat_file(file_name)
  File "/home/dgorbani/software/robotology-superbuild/build/install/lib/python3/dist-packages/robot_log_visualizer/file_reader/signal_provider.py", line 140, in open_mat_file
    self.text_logging_data["log"] = self.__populate_text_logging_data(
  File "/home/dgorbani/software/robotology-superbuild/build/install/lib/python3/dist-packages/robot_log_visualizer/file_reader/signal_provider.py", line 95, in __populate_text_logging_data
    data[key] = self.__populate_text_logging_data(file_object=value)
  File "/home/dgorbani/software/robotology-superbuild/build/install/lib/python3/dist-packages/robot_log_visualizer/file_reader/signal_provider.py", line 95, in __populate_text_logging_data
    data[key] = self.__populate_text_logging_data(file_object=value)
  File "/home/dgorbani/software/robotology-superbuild/build/install/lib/python3/dist-packages/robot_log_visualizer/file_reader/signal_provider.py", line 75, in __populate_text_logging_data
    if text_ref.shape[0] == len(data[key]["timestamps"]):
TypeError: len() of unsized object
```

I can visualize data of matlab structure in which there is not that data field.

GiulioRomualdi, 2023-07-14:
Hi @davidegorbani thank you for reporting! 
Unfortunately I don't have time to fix it before the next two weeks. 

If you want you may open a PR with the fix. In theory it should be enough to catch the exception.

Thank you very much!

davidegorbani, 2023-07-14:
Thank you! I will try to solve it as you suggested.

davidegorbani, 2023-07-18:
Adding [this line](https://github.com/ami-iit/bipedal-locomotion-framework/blob/ec549f6199885fe276766a17f62b1d178deae4e8/devices/YarpRobotLoggerDevice/app/robots/ergoCubSN000/yarp-robot-logger.xml#L20) of code in the `xml` file as suggested by @GiulioRomualdi solved the problem.

GiulioRomualdi, 2023-10-05:
Hi @davidegorbani by any chance do you have a dataset where I can replicate the issue?

davidegorbani, 2023-10-06:
Hello @GiulioRomualdi, here is a dataset that I just acquired that replicates the error.

[dataset.zip](https://github.com/ami-iit/robot-log-visualizer/files/12827948/dataset.zip)

GiulioRomualdi, 2023-10-06:
Fixed by https://github.com/ami-iit/robot-log-visualizer/pull/65

PR #65 (2023-10-06): Fix 57
This fixes #57
closed 2023-10-06 by GiulioRomualdi