gbionics/robot-log-visualizer #58: Problem with superbuild conda-forge instalation

carloscp3009, 2023-07-20:
Hi, 
I installed the robot log visualizer using robotology-superbuild (conda-forge installation on ubuntu 22.04) but the software didnt even initialized.
After re-installing (pyqtwebengine and qtpy) the software launched but it did not work properly.
Similar problem with yarpview (Could not make it work)

Dont know if it is a compatibility issue (with conda-forge installation on ubuntu 22.04) or something with the versions.
Thanks

DanielePucci, 2023-07-21:
Hi @carloscp3009, probably to help the debug you may add the details of the problem you see? Thanks

traversaro, 2023-07-31:
@carloscp3009 are you still experiencing this problem? Can you report the output of the following commands:
* `conda info`
* `mamba list`
* The output when you try to launch `yarpview` or `robot-log-visualizer`

Thanks!

carloscp3009, 2023-08-02:
I deleted the environment and reinstalled and compiled everything again by using apt instead and it worked immediately. I will add all details in plain text next time.

traversaro, 2023-08-02:
Ok, thanks!

traversaro, 2023-10-08:
Perhaps the problem was https://github.com/ami-iit/robot-log-visualizer/issues/67 .

traversaro, 2023-11-15:
@carloscp3009 if you solved the problem, can you close the issue? Thanks!

closed 2023-11-15 by carloscp3009