gbionics/robot-log-visualizer #60: `robot-log-visualizer` doesn't start on Windows

FabioBergonti, 2023-09-28:
I installed `robot-log-visualizer` on a new environment with `conda` but it doesn't work

This is the error

``` shell
You can open the visualizer by visiting the following URL:
http://127.0.0.1:7000/static/
C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\lib\site-packages\robot_log_visualizer\plotter\matplotlib_viewer_canvas.py:45: UserWarning: frames=None which we can infer the length of, did not pass an explicit *save_count* and passed cache_frame_data=True.  To avoid a possibly unbounded cache, frame data caching has been disabled. To suppress this warning either pass `cache_frame_data=False` or `save_count=MAX_FRAMES`.
  self.vertical_line_anim = animation.FuncAnimation(
Traceback (most recent call last):
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\Scripts\robot-log-visualizer-script.py", line 10, in <module>
    sys.exit(main())
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\lib\site-packages\robot_log_visualizer\__main__.py", line 38, in main
    gui = RobotViewerMainWindow(
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\lib\site-packages\robot_log_visualizer\ui\gui.py", line 122, in __init__
    self.toolButton_on_click()
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\lib\site-packages\robot_log_visualizer\ui\gui.py", line 248, in toolButton_on_click
    PlotItem(signal_provider=self.signal_provider, period=self.animation_period)
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\lib\site-packages\robot_log_visualizer\ui\plot_item.py", line 17, in __init__
    self.canvas = MatplotlibViewerCanvas(
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\lib\site-packages\robot_log_visualizer\plotter\matplotlib_viewer_canvas.py", line 45, in __init__
    self.vertical_line_anim = animation.FuncAnimation(
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\lib\site-packages\matplotlib\animation.py", line 1695, in __init__
    super().__init__(fig, **kwargs)
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\lib\site-packages\matplotlib\animation.py", line 1417, in __init__
    super().__init__(fig, event_source=event_source, *args, **kwargs)
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\lib\site-packages\matplotlib\animation.py", line 888, in __init__
    self._setup_blit()
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\lib\site-packages\matplotlib\animation.py", line 1216, in _setup_blit
    self._init_draw()
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\lib\site-packages\matplotlib\animation.py", line 1751, in _init_draw
    self._drawn_artists = self._init_func()
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\lib\site-packages\robot_log_visualizer\plotter\matplotlib_viewer_canvas.py", line 124, in init_vertical_line
    return self.vertical_line, *(self.active_paths.values())
AttributeError: 'MatplotlibViewerCanvas' object has no attribute 'active_paths'
``` 

<details><summary>mamba list</summary>
<p>

```
# packages in environment at C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv:
#
# Name                    Version                   Build  Channel
assimp                    5.2.4                h4dcb625_2    conda-forge
asttokens                 2.4.0              pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.5              pyhd8ed1ab_0    conda-forge
boost-cpp                 1.78.0               h9f4b32c_4    conda-forge
brotli                    1.1.0                hcfcfb64_0    conda-forge
brotli-bin                1.1.0                hcfcfb64_0    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2023.7.22            h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
contourpy                 1.1.1           py310h232114e_1    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
exceptiongroup            1.1.3              pyhd8ed1ab_0    conda-forge
executing                 1.2.0              pyhd8ed1ab_0    conda-forge
fonttools                 4.42.1          py310h8d17308_0    conda-forge
freetype                  2.12.1               hdaf720e_2    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
glib                      2.78.0               h12be248_0    conda-forge
glib-tools                2.78.0               h12be248_0    conda-forge
gst-plugins-base          1.22.6               h001b923_1    conda-forge
[... 139 lines cut ...]
    conda-build version : not installed
         python version : 3.9.16.final.0
       virtual packages : __archspec=1=x86_64
                          __cuda=12.2=0
                          __win=0=0
       base environment : C:\Users\fbergonti\AppData\Local\mambaforge  (writable)
      conda av data dir : C:\Users\fbergonti\AppData\Local\mambaforge\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : C:\Users\fbergonti\AppData\Local\mambaforge\pkgs
                          C:\Users\fbergonti\.conda\pkgs
                          C:\Users\fbergonti\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\fbergonti\AppData\Local\mambaforge\envs
                          C:\Users\fbergonti\.conda\envs
                          C:\Users\fbergonti\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.5.0 requests/2.31.0 CPython/3.9.16 Windows/10 Windows/10.0.22621
          administrator : False
             netrc file : None
           offline mode : False
```

</p>
</details>

GiulioRomualdi, 2023-10-02:
That's strange since `active_paths` is defined in `MatplotlibViewerCanvas`

https://github.com/ami-iit/robot-log-visualizer/blob/02621ebaab8af7ad09386bbf2969616c147e2b09/robot_log_visualizer/plotter/matplotlib_viewer_canvas.py#L44-L45

FabioBergonti, 2023-10-03:
The last version of `robot-log-visualizer` available on https://anaconda.org/robotology/robot-log-visualizer is `v0.2.2` (Last upload: 1 year and 1 month ago)

cc @traversaro

traversaro, 2023-10-03:
But for some reason, `robot-log-visualizer` is not uploaded, as one can check by looking at the lines that start with "Uploading file".

traversaro, 2023-10-03:
This is a regression of https://github.com/robotology/robotology-superbuild/pull/1243 . I converted the recipe to noarch, but I did not checked that noarch packages were being uploaded correctly.

traversaro, 2023-10-04:
@FabioBergonti an updated version of the package was uploaded last night: https://anaconda.org/robotology/robot-log-visualizer/files . Can you try if it fixes your issue?

FabioBergonti, 2023-10-04:
Thanks @traversaro 

v0.6.1 is now installed, but when I run the visualizer it can't import `icub_models`

```
(rlv) C:\Users\fbergonti>robot-log-visualizer
Traceback (most recent call last):
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\Scripts\robot-log-visualizer-script.py", line 5, in <module>
    from robot_log_visualizer.__main__ import main
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\Lib\site-packages\robot_log_visualizer\__main__.py", line 17, in <module>
    from robot_log_visualizer.robot_visualizer.meshcat_provider import MeshcatProvider
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv\Lib\site-packages\robot_log_visualizer\robot_visualizer\meshcat_provider.py", line 7, in <module>
    import icub_models
ModuleNotFoundError: No module named 'icub_models'
```

<details><summary>mamba list</summary>
<p>

```
(rlv) C:\Users\fbergonti>mamba list
# packages in environment at C:\Users\fbergonti\AppData\Local\mambaforge\envs\rlv:
#
# Name                    Version                   Build  Channel
assimp                    5.3.1                h81f0834_2    conda-forge
asttokens                 2.4.0              pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.5              pyhd8ed1ab_0    conda-forge
brotli                    1.1.0                hcfcfb64_1    conda-forge
brotli-bin                1.1.0                hcfcfb64_1    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2023.7.22            h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
contourpy                 1.1.1           py311h005e61a_1    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
exceptiongroup            1.1.3              pyhd8ed1ab_0    conda-forge
executing                 1.2.0              pyhd8ed1ab_0    conda-forge
fonttools                 4.43.0          py311ha68e1ae_0    conda-forge
freetype                  2.12.1               hdaf720e_2    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
glfw                      3.3.8                hcfcfb64_1    conda-forge
glib                      2.78.0               h12be248_0    conda-forge
glib-tools                2.78.0               h12be248_0    conda-forge
gst-plugins-base          1.22.6               h001b923_2    conda-forge
gstreamer                 1.22.6               hb4038d2_2    conda-forge
h5py                      3.9.0           nompi_py311h0d04526_103    conda-forge
hdf5                      1.14.2          nompi_h73e8ff5_100    conda-forge
icu                       73.2                 h63175ca_0    conda-forge
icub-models               1.23.0              h76494f8_47    robotology
idyntree                  9.1.0           py311h0a7ed4b_3    conda-forge
intel-openmp              2023.2.0         h57928b3_50496    conda-forge
ipopt                     3.14.12              ha9547d1_1    conda-forge
ipython                   8.16.1             pyh5737063_0    conda-forge
irrlicht                  1.8.5                h6ba1e2c_3    conda-forge
jedi                      0.19.1             pyhd8ed1ab_0    conda-forge
kiwisolver                1.4.5           py311h005e61a_1    conda-forge
krb5                      1.21.2               heb0366b_0    conda-forge
lcms2                     2.15                 he9d350c_2    conda-forge
lerc                      4.0.0                h63175ca_0    conda-forge
libaec                    1.1.1                h63175ca_0    conda-forge
libblas                   3.9.0              18_win64_mkl    conda-forge
libboost                  1.82.0               h65993cd_5    conda-forge
libbrotlicommon           1.1.0                hcfcfb64_1    conda-forge
libbrotlidec              1.1.0                hcfcfb64_1    conda-forge
[... 83 lines cut ...]
tk                        8.6.13               hcfcfb64_0    conda-forge
toml                      0.10.2             pyhd8ed1ab_0    conda-forge
tomli                     2.0.1              pyhd8ed1ab_0    conda-forge
tornado                   6.3.3           py311ha68e1ae_1    conda-forge
traitlets                 5.11.2             pyhd8ed1ab_0    conda-forge
typing_extensions         4.8.0              pyha770c72_0    conda-forge
tzdata                    2023c                h71feb2d_0    conda-forge
u-msgpack-python          2.8.0              pyhd8ed1ab_0    conda-forge
ucrt                      10.0.22621.0         h57928b3_0    conda-forge
vc                        14.3                h64f974e_17    conda-forge
vc14_runtime              14.36.32532         hdcecf7f_17    conda-forge
vs2015_runtime            14.36.32532         h05e6639_17    conda-forge
wcwidth                   0.2.8              pyhd8ed1ab_0    conda-forge
wheel                     0.41.2             pyhd8ed1ab_0    conda-forge
[... cut ...] 1               hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zlib                      1.2.13               hcfcfb64_5    conda-forge
zstd                      1.5.5                h12be248_0    conda-forge
```

</p>
</details>

FabioBergonti, 2023-10-04:
also icub models is quite old https://anaconda.org/robotology/icub-models
`v1.23.0` (Last upload: 1 year and 6 months ago)

FabioBergonti, 2023-10-04:
Installing adding conda-forge as channel solved the [previous problem](https://github.com/ami-iit/robot-log-visualizer/issues/60#issuecomment-1746359165)
`mamba install -c conda-forge -c robotology robot-log-visualizer`
Now the `robot-log-visualizer` opens, I will soon test it with a data

traversaro, 2023-10-04:
> also icub models is quite old on robotology https://anaconda.org/robotology/icub-models `v1.23.0` (Last upload: 1 year and 6 months ago) I will try to install it on conda-forge

Yes, that is expected. As we migrated packages over time from robotology to conda-forge (see https://github.com/robotology/robotology-superbuild/issues/752), so conda-forge must always have higher priority w.r.t. to robotology channel. This is documented in https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md#install-robotology-packages, but probably we can make it more clear.

FabioBergonti, 2023-10-04:
> I will test it soon with a data

It works! Thanks @GiulioRomualdi and @traversaro 
Closing!

closed 2023-10-04 by FabioBergonti

traversaro, 2023-10-08:
@FabioBergonti are you able to visualize the 3D model and the video? For the video probably `mamba install gst-plugins-good gst-plugins-bad` could help, see https://github.com/ami-iit/robot-log-visualizer/issues/67 . However, on my test on Windows I still can't visualize the 3D model.

FabioBergonti, 2023-10-08:
Yes, I was able to visualize the 3D model of ironcub
The video I didn't try

FabioBergonti, 2024-02-21:
> The video I didn't try

Hey @traversaro 
Today I tried the video and it didn't work

```
(ironcub-dev) C:\Users\fbergonti\Desktop\tes>robot-log-visualizer
You can open the visualizer by visiting the following URL:
http://127.0.0.1:7000/static/
C:\Users\fbergonti\AppData\Local\mambaforge\envs\ironcub-dev\Lib\site-packages\robot_log_visualizer\plotter\matplotlib_viewer_canvas.py:47: UserWarning: frames=None which we can infer the length of, did not pass an explicit *save_count* and passed cache_frame_data=True.  To avoid a possibly unbounded cache, frame data caching has been disabled. To suppress this warning either pass `cache_frame_data=False` or `save_count=MAX_FRAMES`.
  self.vertical_line_anim = animation.FuncAnimation(
DirectShowPlayerService::doRender: Unresolved error code 0x80040266 (IDispatch error #102)
DirectShowPlayerService::doRender: Unresolved error code 0x80040266 (IDispatch error #102)
Traceback (most recent call last):
  File "C:\Users\fbergonti\AppData\Local\mambaforge\envs\ironcub-dev\Lib\site-packages\matplotlib\backend_bases.py", line 1226, in _on_timer
    def _on_timer(self):
```

I have the `gst-plugins-good` and `gst-plugins-bad`

<details><summary>mamba list</summary>
<p>

``` 
(ironcub-dev) C:\Users\fbergonti\Desktop\tes>mamba list
# packages in environment at C:\Users\fbergonti\AppData\Local\mambaforge\envs\ironcub-dev:
#
# Name                    Version                   Build  Channel
ace                       7.1.3                h63175ca_2    conda-forge
aiohttp                   3.9.3           py311ha68e1ae_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.8.1                h63175ca_0    conda-forge
assimp                    5.3.1                h81f0834_2    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
azure-core-cpp            1.10.3               h249a519_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h91493d7_0    conda-forge
azure-storage-common-cpp  12.5.0               h91493d7_2    conda-forge
bipedal-locomotion-framework 0.18.0               h57928b3_1    conda-forge
bipedal-locomotion-framework-python 0.18.0          py311h4533acf_1    conda-forge
blockfactory              0.8.5              h63175ca_116    robotology
blosc                     1.21.5               hdccc3a2_0    conda-forge
brotli                    1.1.0                hcfcfb64_1    conda-forge
brotli-bin                1.1.0                hcfcfb64_1    conda-forge
brotli-python             1.1.0           py311h12c1d0e_1    conda-forge
bullet-cpp                3.25                 hf63dbb6_2    conda-forge
bzip2                     1.0.8                hcfcfb64_5    conda-forge
c-ares                    1.26.0               hcfcfb64_0    conda-forge
c-compiler                1.7.0                hcfcfb64_0    conda-forge
ca-certificates           2024.2.2             h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               h1fef639_0    conda-forge
casadi                    3.6.3           py311h4930016_5    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cfitsio                   4.3.1                h9b0cee5_0    conda-forge
charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.28.3               hf0feee3_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h1ea47a8_1    conda-forge
colcon-core               0.15.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.5              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-notification       0.2.15          py311h1ea47a8_3    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
[... 356 lines cut ...]
xerces-c                  3.2.5                h63175ca_0    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.1.1                hcd874cb_0    conda-forge
xorg-libsm                1.2.4                hcd874cb_0    conda-forge
xorg-libx11               1.8.7                hefa74cf_0    conda-forge
xorg-libxau               1.0.11               hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xorg-libxext              1.3.4                hcd874cb_2    conda-forge
xorg-libxpm               3.5.17               hcd874cb_0    conda-forge
xorg-libxt                1.3.0                hcd874cb_1    conda-forge
xorg-xextproto            7.3.0             hcd874cb_1003    conda-forge
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
[... cut ...]
yarl                      1.9.4           py311ha68e1ae_0    conda-forge
ycm-cmake-modules         0.16.2               h63175ca_0    conda-forge
zeromq                    4.3.5                h63175ca_0    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hcfcfb64_5    conda-forge
zstd                      1.5.5                h12be248_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
```

</p>
</details>

traversaro, 2024-02-21:
Can you try by setting `set LIBVA_DRIVER_NAME=fakedriver` env variable in the terminal before launching `robot-log-visualizer`?

FabioBergonti, 2024-02-21:
> Can you try by setting `set LIBVA_DRIVER_NAME=fakedriver` env variable in the terminal before launching `robot-log-visualizer`?

Done, but it didn't fix the problem

> Also ,can you report `gst-inspect-1.0 --print-all` ?

Here it is
[gst-inspect.txt](https://github.com/ami-iit/robot-log-visualizer/files/14363447/gst-inspect.txt)