gbionics/robot-log-visualizer #61: `robot-log-visualizer` doesn't start on WSL2

FabioBergonti, 2023-09-28:
I installed `robot-log-visualizer` via robolotogy_superbuild (source)

This is the error

```
You can open the visualizer by visiting the following URL:
http://127.0.0.1:7006/static/
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
qt.svg: Invalid path data; path truncated.
D3D12: Removing Device.
Segmentation fault
```

traversaro, 2023-10-08:
> I installed `robot-log-visualizer` via robolotogy_superbuild (source) and pip but it doesn't start

If you install `robot-log-visualizer` via the robotology-superbuild, you should not install it via pip. Or you first tried with robotology-superbuild and then with pip?

FabioBergonti, 2023-10-08:
> Or you first tried with robotology-superbuild and then with pip?

Yeah, I first tried with robotology and then with pip
I did not try with conda

traversaro, 2023-10-08:
pip on the top of the robotology-superbuild or in a clean environment?

FabioBergonti, 2023-10-08:
> pip on the top of the robotology-superbuild or in a clean environment?

on a clean environment

<details><summary>pip list</summary>
<p>

``` 
Package              Version
-------------------- -------
asttokens            2.4.0
backcall             0.2.0
contourpy            1.1.1
cycler               0.12.1
decorator            5.1.1
exceptiongroup       1.1.3
executing            2.0.0
fonttools            4.43.1
h5py                 3.9.0
icub-models          2.3.0
idyntree             9.1.0
ipython              8.16.1
jedi                 0.19.1
kiwisolver           1.4.5
matplotlib           3.8.0
matplotlib-inline    0.1.6
meshcat              0.3.2
numpy                1.26.0
packaging            23.2
parso                0.8.3
pexpect              4.8.0
pickleshare          0.7.5
Pillow               10.0.1
pip                  22.0.2
prompt-toolkit       3.0.39
ptyprocess           0.7.0
pure-eval            0.2.2
Pygments             2.16.1
pyngrok              7.0.0
pyparsing            3.1.1
PyQt5                5.15.9
PyQt5-Qt5            5.15.2
PyQt5-sip            12.12.2
pyqtconsole          1.2.3
PyQtWebEngine        5.15.6
PyQtWebEngine-Qt5    5.15.2
python-dateutil      2.8.2
PyYAML               6.0.1
pyzmq                25.1.1
QtPy                 2.4.0
robot-log-visualizer 0.6.1
setuptools           59.6.0
six                  1.16.0
stack-data           0.6.3
tornado              6.3.3
traitlets            5.11.2
u-msgpack-python     2.8.0
wcwidth              0.2.8
``` 

</p>
</details>

FabioBergonti, 2023-10-08:
same error with `robot-log-visualizer` installed with mamba

```
(rlv) fbergonti@IITICUBLAP279:~$ robot-log-visualizer
You can open the visualizer by visiting the following URL:
http://127.0.0.1:7002/static/
/home/fbergonti/mambaforge/envs/rlv/lib/python3.11/site-packages/robot_log_visualizer/plotter/matplotlib_viewer_canvas.py:47: UserWarning: frames=None which we can infer the length of, did not pass an explicit *save_count* and passed cache_frame_data=True.  To avoid a possibly unbounded cache, frame data caching has been disabled. To suppress this warning either pass `cache_frame_data=False` or `save_count=MAX_FRAMES`.
  self.vertical_line_anim = animation.FuncAnimation(
D3D12: Removing Device.
Segmentation fault
```

<details><summary>conda list</summary>
<p>

```
(rlv) fbergonti@IITICUBLAP279:~$ conda list
# packages in environment at /home/fbergonti/mambaforge/envs/rlv:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
alsa-lib                  1.2.10               hd590300_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
asttokens                 2.4.0              pyhd8ed1ab_0    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.5              pyhd8ed1ab_0    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.19.1               hd590300_0    conda-forge
ca-certificates           2023.7.22            hbcca054_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.16.0            h0c91306_1017    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
contourpy                 1.1.1           py311h9547e67_1    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
exceptiongroup            1.1.3              pyhd8ed1ab_0    conda-forge
executing                 1.2.0              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               h14ed4e7_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.43.1          py311h459d7ec_0    conda-forge
freetype                  2.12.1               h267a509_2    conda-forge
gettext                   0.21.1               h27087fc_0    conda-forge
giflib                    5.2.1                h0b41bf4_3    conda-forge
glfw                      3.3.8                hd590300_1    conda-forge
glib                      2.78.0               hfc55251_0    conda-forge
glib-tools                2.78.0               hfc55251_0    conda-forge
graphite2                 1.3.13            h58526e2_1001    conda-forge
gst-plugins-base          1.22.6               h8e1006c_2    conda-forge
gstreamer                 1.22.6               h98fc4e7_2    conda-forge
h5py                      3.9.0           nompi_py311h3839ddf_103    conda-forge
harfbuzz                  8.2.1                h3d44ed6_0    conda-forge
hdf5                      1.14.2          nompi_h4f84152_100    conda-forge
icu                       73.2                 h59595ed_0    conda-forge
icub-models               2.3.0           py311hb755f60_1    conda-forge
idyntree                  9.1.0           py311hd1120ad_3    conda-forge
[... 154 lines cut ...]
xorg-libxdamage           1.1.5                h7f98852_1    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h0b41bf4_2    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxinerama          1.1.5                h27087fc_0    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.11               hd590300_0    conda-forge
xorg-libxtst              1.2.3             h7f98852_1002    conda-forge
xorg-randrproto           1.5.0             h7f98852_1001    conda-forge
xorg-recordproto          1.14.2            h7f98852_1002    conda-forge
[... cut ...]
xorg-xf86vidmodeproto     2.3.1             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge
```

</p>
</details>

traversaro, 2023-10-08:
Strange, for me on WSL2 it is working, this is the environment:
~~~
(rlvtest) traversaro@IITICUBLAP257:~$ mamba list
# packages in environment at /home/traversaro/miniforge3/envs/rlvtest:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
alsa-lib                  1.2.10               hd590300_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
asttokens                 2.4.0              pyhd8ed1ab_0    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.5              pyhd8ed1ab_0    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.19.1               hd590300_0    conda-forge
ca-certificates           2023.7.22            hbcca054_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.16.0            h0c91306_1017    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
contourpy                 1.1.1           py311h9547e67_1    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
exceptiongroup            1.1.3              pyhd8ed1ab_0    conda-forge
executing                 1.2.0              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               h14ed4e7_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.43.1          py311h459d7ec_0    conda-forge
freetype                  2.12.1               h267a509_2    conda-forge
gettext                   0.21.1               h27087fc_0    conda-forge
giflib                    5.2.1                h0b41bf4_3    conda-forge
glfw                      3.3.8                hd590300_1    conda-forge
glib                      2.78.0               hfc55251_0    conda-forge
glib-tools                2.78.0               hfc55251_0    conda-forge
graphite2                 1.3.13            h58526e2_1001    conda-forge
gst-plugins-base          1.22.6               h8e1006c_2    conda-forge
gstreamer                 1.22.6               h98fc4e7_2    conda-forge
h5py                      3.9.0           nompi_py311h3839ddf_103    conda-forge
harfbuzz                  8.2.1                h3d44ed6_0    conda-forge
hdf5                      1.14.2          nompi_h4f84152_100    conda-forge
icu                       73.2                 h59595ed_0    conda-forge
icub-models               2.3.0           py311hb755f60_1    conda-forge
idyntree                  9.1.0           py311hd1120ad_3    conda-forge
ipopt                     3.14.12              he6d3896_1    conda-forge
ipython                   8.16.1             pyh0d859eb_0    conda-forge
irrlicht                  1.8.5                h3749e40_3    conda-forge
jedi                      0.19.1             pyhd8ed1ab_0    conda-forge
keyutils                  1.6.1                h166bdaf_0    conda-forge
kiwisolver                1.4.5           py311h9547e67_1    conda-forge
krb5                      1.21.2               h659d440_0    conda-forge
lame                      3.100             h166bdaf_1003    conda-forge
lcms2                     2.15                 h7f713cb_2    conda-forge
ld_impl_linux-64          2.40                 h41732ed_0    conda-forge
lerc                      4.0.0                h27087fc_0    conda-forge
libaec                    1.1.2                h59595ed_1    conda-forge
libblas                   3.9.0           18_linux64_openblas    conda-forge
libboost                  1.82.0               h6fcfa73_6    conda-forge
[... 137 lines cut ...]
xorg-libxau               1.0.11               hd590300_0    conda-forge
xorg-libxcomposite        0.4.6                h0b41bf4_1    conda-forge
xorg-libxcursor           1.2.0                h0b41bf4_1    conda-forge
xorg-libxdamage           1.1.5                h7f98852_1    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h0b41bf4_2    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
[... cut ...]
xorg-xf86vidmodeproto     2.3.1             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge
~~~

traversaro, 2023-10-08:
That: 
~~~
D3D12: Removing Device.
Segmentation fault
~~~

suggests that something is wrong at the OpenGL level. Can you try to set `export LIBGL_ALWAYS_SOFTWARE=true` and try if the gui loads correctly? Perhaps do you have something else of strange in that WSL2 install?

FabioBergonti, 2023-10-08:
> Can you try to set `export LIBGL_ALWAYS_SOFTWARE=true` and try if the gui loads correctly? Perhaps do you have something else of strange in that WSL2 install? 

setting `export LIBGL_ALWAYS_SOFTWARE=true` the gui opens  🎉

[image]

But I cannot see the robot and I got this output

```
(rlv) fbergonti@IITICUBLAP279:~$ robot-log-visualizer
You can open the visualizer by visiting the following URL:
http://127.0.0.1:7005/static/
/home/fbergonti/code/robotology-superbuild/build/install/lib/python3/dist-packages/robot_log_visualizer/plotter/matplotlib_viewer_canvas.py:47: UserWarning: frames=None which we can infer the length of, did not pass an explicit *save_count* and passed cache_frame_data=True.  To avoid a possibly unbounded cache, frame data caching has been disabled. To suppress this warning either pass `cache_frame_data=False` or `save_count=MAX_FRAMES`.
  self.vertical_line_anim = animation.FuncAnimation(
[11721:7:1008/195611.514463:ERROR:command_buffer_proxy_impl.cc(141)] ContextResult::kTransientFailure: Failed to send GpuChannelMsg_CreateCommandBuffer.
[11612:11716:1008/195611.700235:ERROR:context_group.cc(154)] ContextResult::kFatalFailure: WebGL2 blocklisted
[11612:11716:1008/195611.710493:ERROR:context_group.cc(146)] ContextResult::kFatalFailure: WebGL1 blocklisted
[11612:11716:1008/195611.722306:ERROR:context_group.cc(146)] ContextResult::kFatalFailure: WebGL1 blocklisted
[11612:11716:1008/195611.731002:ERROR:context_group.cc(154)] ContextResult::kFatalFailure: WebGL2 blocklisted
[11612:11716:1008/195611.741454:ERROR:context_group.cc(146)] ContextResult::kFatalFailure: WebGL1 blocklisted
[11612:11716:1008/195611.750000:ERROR:context_group.cc(146)] ContextResult::kFatalFailure: WebGL1 blocklisted
js: THREE.WebGLRenderer: Error creating WebGL context.
js: Uncaught Error: Error creating WebGL context.
/home/fbergonti/code/robotology-superbuild/build/install/lib/python3/dist-packages/robot_log_visualizer/plotter/matplotlib_viewer_canvas.py:107: UserWarning: frames=None which we can infer the length of, did not pass an explicit *save_count* and passed cache_frame_data=True.  To avoid a possibly unbounded cache, frame data caching has been disabled. To suppress this warning either pass `cache_frame_data=False` or `save_count=MAX_FRAMES`.
  self.vertical_line_anim = animation.FuncAnimation(

```

> If you do not set `export LIBGL_ALWAYS_SOFTWARE=true` , does `glxgears` work?

Yes, it works

FabioBergonti, 2023-10-08:
> But I cannot see the robot

I can see it typing `http://127.0.0.1:7005/static/` in chrome

[image]