gbionics/robot-log-visualizer #67: robot-log-visualizer installed by conda-forge cannot open .mp4 generated by OpenCV's VideoWriter

traversaro, 2023-10-08:
The file is:
~~~
traversaro@IITICUBLAP257:~/datatsets/datasets/robot_logger_device_2023_10_06_12_35_58$ file robot_logger_devic
e_2023_10_06_12_35_58_logitech_rgb.mp4
robot_logger_device_2023_10_06_12_35_58_logitech_rgb.mp4: ISO Media, MP4 Base Media v1 [ISO 14496-12:2003]
traversaro@IITICUBLAP257:~/datatsets/datasets/robot_logger_device_2023_10_06_12_35_58$ ffprobe robot_logger_device_2023_10_06_12_35_58_logitech_rgb.mp4
ffprobe version 4.4.2-0ubuntu0.22.04.1 Copyright (c) 2007-2021 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.2.0-19ubuntu1)
  configuration: --prefix=/usr --extra-version=0ubuntu0.22.04.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-pocketsphinx --enable-librsvg --enable-libmfx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'robot_logger_device_2023_10_06_12_35_58_logitech_rgb.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2mp41
    encoder         : Lavf58.76.100
  Duration: 00:03:10.10, start: 0.000000, bitrate: 1680 kb/s
  Stream #0:0(und): Video: mpeg4 (Simple Profile) (mp4v / 0x7634706D), yuv420p, 640x480 [SAR 1:1 DAR 4:3], 1680 kb/s, 10 fps, 10 tbr, 10240 tbn, 10 tbc (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
traversaro@IITICUBLAP257:~/datatsets/datasets/robot_logger_device_2023_10_06_12_35_58$ gst-typefind-1.0 robot_logger_device_2023_10_06_12_35_58_logitech_rgb.mp4
robot_logger_device_2023_10_06_12_35_58_logitech_rgb.mp4 - video/quicktime, variant=(string)iso
~~~

The code that created the file is:
https://github.com/ami-iit/bipedal-locomotion-framework/blob/7723da13e4ddab9ebafe53df1daacf2e04719126/devices/YarpRobotLoggerDevice/src/YarpRobotLoggerDevice.cpp#L907

mamba list:
~~~
# packages in environment at /home/traversaro/miniforge3/envs/rlv:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
alsa-lib                  1.2.10               hd590300_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
asttokens                 2.4.0              pyhd8ed1ab_0    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.5              pyhd8ed1ab_0    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.19.1               hd590300_0    conda-forge
ca-certificates           2023.7.22            hbcca054_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.16.0            h0c91306_1017    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
contourpy                 1.1.1           py311h9547e67_1    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
exceptiongroup            1.1.3              pyhd8ed1ab_0    conda-forge
executing                 1.2.0              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
[... 189 lines cut ...]
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge
~~~

To decouple a bit the issue, I tried to play the file with `gst-play-1.0` (as anyhow qtmultimedia uses gstreamer under the hood) and this is the error:
~~~
(rlv) traversaro@IITICUBLAP257:~/datatsets/datasets/robot_logger_device_2023_10_06_12_35_58$ gst-play-1.0 robo
t_logger_device_2023_10_06_12_35_58_logitech_rgb.mp4
Press 'k' to see a list of keyboard shortcuts.
Now playing /home/traversaro/datatsets/datasets/robot_logger_device_2023_10_06_12_35_58/robot_logger_device_2023_10_06_12_35_58_logitech_rgb.mp4
WARNING No decoder available for type 'video/quicktime, variant=(string)iso'.
WARNING debug information: ../gst/playback/gsturidecodebin.c(982): unknown_type_cb (): /GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0
ERROR Your GStreamer installation is missing a plug-in. for file:///home/traversaro/datatsets/datasets/robot_logger_device_2023_10_06_12_35_58/robot_logger_device_2023_10_06_12_35_58_logitech_rgb.mp4
ERROR debug information: ../gst/playback/gsturidecodebin.c(1049): no_more_pads_full (): /GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0:
no suitable plugins found:
../gst/playback/gstdecodebin2.c(4704): gst_decode_bin_expose (): /GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0:
no suitable plugins found:
Missing decoder: Quicktime (video/quicktime, variant=(string)iso)

Reached end of play list.
~~~

traversaro, 2023-10-08:
The video plays correctly by installing:
~~~
mamba install gst-plugins-good gst-plugins-bad
~~~
so I proposed to at least install them as run dependencies of the the robot-log-visualizer conda package. We should consider if it make sense to add them to robotology-superbuild from source dependencies (I guess we also have a similar problem with apt).

gabrielenava, 2023-10-11:
>The video plays correctly by installing:

I have activated the mamba environment and I have run the command in https://github.com/ami-iit/robot-log-visualizer/issues/67#issuecomment-1752102250. it installed only `gst-plugins-bad` (I guess the other plugin was already there). Still, the video did not work on the experiment laptop: same problem as before

traversaro, 2023-10-11:
(Also, can I pass to the roof to debug?)

gabrielenava, 2023-10-11:
>(Also, can I pass to the roof to debug?)

we were debugging another issue, but tomorrow we will go again on the roof and we'll be available for debugging the visualizer!

gabrielenava, 2023-10-13:
in the end because of anther issue (https://github.com/ami-iit/robot-log-visualizer/issues/56#issuecomment-1761822471) I have deleted the environment, thinking that it might be related to the problem I was experiencing. It turned out that the problem was something else so I can re-install the environment and try again next week.

traversaro, 2023-11-15:
As confirmed by @akhilsathuluri in https://github.com/ami-iit/robot-log-visualizer/issues/56#issuecomment-1809924714, `mamba install gst-plugins-good gst-plugins-bad` fixes the problems. Let's implement that solution, see https://github.com/robotology/robotology-superbuild/pull/1520 .

closed 2023-11-15 by traversaro