gbionics/robot-log-visualizer #71: Robot visualization non syncronized with data

carloscp3009, 2023-11-14:
The data plot in the visualizer is not syncronized in the time axes with the motion of the robot visualizer. Attached there is a video that shows the signal of the knee that match the motion of the robot in the jabra but not in the visualizer. 
Note: In the video, the time axes was already modify with the intention of fixing the problem, in the original configuration the robot visualizer only move the fingers.
There is also attached the data used in the video. 

[Screencast from 14-11-2023 19:44:19.webm](https://github.com/ami-iit/robot-log-visualizer/assets/44415073/115f095a-3b36-44ea-b996-e5db6ace688f)

https://istitutoitalianotecnologia-my.sharepoint.com/:u:/g/personal/carlos_cardenas_iit_it/EY4zNd52uiFGqd5npTz7qaABj57lbA1MOIjUSBFbGp5wmg?e=bpes7g

DanielePucci, 2023-11-14:
CC @GiulioRomualdi @paolo-viceconte

GiulioRomualdi, 2023-11-15:
Thank you, @carloscp3009, for reporting! I think the issue was there since the beginning. If you want you canctry to fix the problem, otherwise I'll try to give it a look in the week.

GiulioRomualdi, 2023-11-15:
Fixed by https://github.com/ami-iit/robot-log-visualizer/pull/72

carloscp3009, 2023-11-15:
Thanks @GiulioRomualdi for your help. I tested and it seems to be working right!

closed 2023-11-15 by carloscp3009