gbionics/robot-log-visualizer #78: Selecting multiple signal by swiping does not update plot

giotherobot, 2024-02-02:
I was trying to select multiple signals at the same time by dragging the mouse on the signal list. This works in the sense that I can see the selection happen in the signal list (the signals get highlited), but the plot does not seem to update until a single signal gets selected/deselected.

S-Dafarra, 2024-02-13:
I wonder whether the signal is sent by Qt in this case.

You can try using ``itemSelectionChanged`` instead of ``itemClicked`` in https://github.com/ami-iit/robot-log-visualizer/blob/996e0d4f22aa48eb2a7b41dc9133115f9494a8f1/robot_log_visualizer/ui/gui.py#L201

Reference doc: https://doc.qt.io/qt-6/qtreewidget.html#itemSelectionChanged

giotherobot, 2024-02-16:
Sadly it is not as easy as a simple signal swap, I tried but the selection mechanism breaks when there are multiple plots.

With a single plot it works fine.