gbionics/robot-log-visualizer #82: Add the possibility to save and load plot presents

GiulioRomualdi, 2024-04-29:
It would be nice to have the possibility to save and load plot presets. I imagine that in the future, it would be possible to run the visualizer with:
```console
robot-log-visualizer --present <present_file>
```
or even better load it from the GUI.

All the plots from the associated dataset would then be automatically displayed. This would save time, as during debugging, we often need to open the same plot for different experiments.

cc @traversaro @S-Dafarra

S-Dafarra, 2024-04-30:
One possibility could be to use QT settings files too: https://doc.qt.io/qt-6/qsettings.html#

S-Dafarra, 2024-04-30:
> One possibility could be to use QT settings files too: https://doc.qt.io/qt-6/qsettings.html#

The good thing about this is that you can edit the file from code too. So you can save/store configurations more easily than classical ``yarp`` ini files.

GiulioRomualdi, 2024-09-26:
cc @FabioBergonti