gbionics/robot-log-visualizer #87: Unable to launch robot-log visualiser due to iDynTree undefined symbol

apicis, 2024-10-18:
Hi, I installed the robot-log-visualizer using the same environment that I used for the robotology-superbuild. I used the command `conda install -c robotology robot-log-visualizer`.

When I run `robot-log-visualizer`, the terminal displays the error:

``` 
ImportError: /media/tapicella/Data/code/robotology-superbuild/build/install/lib/python3.12/site-packages/idyntree/_iDynTree.so: undefined symbol: _ZN8iDynTree53moveLinkFramesToBeCompatibleWithURDFWithGivenBaseLinkERKNS_5ModelERS0_
``` 

Can you help me solve this error, please?

traversaro, 2024-10-18:
Hello, `robot-log-visualizer` is already installed by the superbuild, so you should either use the one provided by the superbuild, or use a different environment.

If you install `robot-log-visualizer` with both the robotology-superbuild and the conda environment of the robotology-superbuild dependencies, you end up having two different iDynTree versions that may have different symbols (one installed by conda, and one installed by the superbuild) and I think this explains your error.

apicis, 2024-10-18:
I uninstalled the robot-log-visualizer using the conda command and now it works. 

Thanks

closed 2024-10-18 by apicis

traversaro, 2024-10-18:
> I uninstalled the robot-log-visualizer using the conda command and now it works. 
> 
> Thanks

Make sure that in the process you also uninstalled all the dependencies installed by robot-log-visualizer that are already installed by the superbuild (in particolar idyntree) otherwise you could have further problems.