gbionics/robot-log-visualizer #89: Unable to launch robot-log-visualizer from robotology-superbuild in Ubuntu 24.04

xela-95, 2024-11-29:
Hi, I installed `robot-log-visualizer` via robotology-superbuild but when I launch it form the terminal I get the following error:

```bash
robot-log-visualizer                                                                                                                      
Traceback (most recent call last):
  File "/home/acroci/repos/robotology-superbuild/build/install/lib/python3.12/site-packages/bin/robot-log-visualizer", line 5, in <module>
    from robot_log_visualizer.__main__ import main
ModuleNotFoundError: No module named 'robot_log_visualizer'

```

When compiling the superbuild everything seems to compile successfully, for example for the robot-log-visualizer the logs are:
```
[110/157] Performing install step for 'robot-log-visualizer'
Processing /home/acroci/repos/robotology-superbuild/src/robot-log-visualizer
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Building wheels for collected packages: robot-log-visualizer
  Building wheel for robot-log-visualizer (pyproject.toml): started
  Building wheel for robot-log-visualizer (pyproject.toml): finished with status 'done'
  Created wheel for robot-log-visualizer: filename=robot_log_visualizer-0.8.1.dev4-py3-none-any.whl size=120459 sha256=51af22fbe5c8b9ac3fa080d3e2d664b72ca21bdb4512fc74c0bda4f483339873
  Stored in directory: /home/acroci/.cache/pip/wheels/16/b6/36/8a6bf32725d111109a632878e56b5f264a7b3f51051781a425
Successfully built robot-log-visualizer
Installing collected packages: robot-log-visualizer
Successfully installed robot-log-visualizer-0.8.1.dev4

```

traversaro, 2024-11-29:
My bet is that there is some confusion between two different Python interpreters in the system.

The superbuild is installed with apt or conda dependencies?

If it is not a binary file, can you report the content of the `/home/acroci/repos/robotology-superbuild/build/install/lib/python3.12/site-packages/bin/robot-log-visualizer` file?

xela-95, 2024-11-29:
> My bet is that there is some confusion between two different Python interpreters in the system.
> 
> The superbuild is installed with apt or conda dependencies?

Using conda dependencies

> If it is not a binary file, can you report the content of the `/home/acroci/repos/robotology-superbuild/build/install/lib/python3.12/site-packages/bin/robot-log-visualizer` file?

Sure, here it is:
```python
#!/home/acroci/mambaforge/envs/iron_rs/bin/python3.12
# -*- coding: utf-8 -*-
import re
import sys
from robot_log_visualizer.__main__ import main
if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw|\.exe)?$', '', sys.argv[0])
    sys.exit(main())

```

xela-95, 2024-11-29:
Thanks to @nicktrem we investigated further and discovered what was happening.

In my setup (related to iRonCub) I have a conda environment with an activation scripts in which are sourced 2  bash setup scripts, one for the superbuild, which among the other things set the PYTHONPATH and the other for iRonCub related software. This one was overriding this env variable setting another python executable, which didn't find all the libraries installed through robotology. 

In the end it was what @traversaro was thinking (as it happens every time :D ).

closed 2024-11-29 by xela-95