gbionics/robot-log-visualizer #90: Catch exception when signal has 1 sample

dariosortino, 2024-12-13:
The application can't load the mat file if it has a signal formed just from 1 sample.

Context: in iFeelBAF Application we also log in the mat file the timestamp in which we perform the calibration, hence that is the 1 sample signal.
As a reference: https://github.com/ami-iit/component_ifeel/issues/1924

dariosortino, 2024-12-13:
I cannot create a branch and open a PR from there. Anyway, the fix is to change the following lines:

https://github.com/ami-iit/robot-log-visualizer/blob/5d3df3de4ea8b78cf1d4303ab75217bdfc809f05/robot_log_visualizer/file_reader/signal_provider.py#L129-L130

with

```python
data[key]["data"] = np.atleast_1d(np.squeeze(np.array(value["data"])))
data[key]["timestamps"] = np.atleast_1d(np.squeeze(np.array(value["timestamps"])))
```

dariosortino, 2024-12-18:
@claudia-lat FYI

Also, forgot to tag @GiulioRomualdi . Can you deal with the PR with this fix?

GiulioRomualdi, 2024-12-18:
Here the trace of the complete error
```
  File "/home/gromualdi/robot-code/robotology-superbuild/build/install/lib/python3/dist-packages/robot_log_visualizer/ui/gui.py", line 690, in dropEvent
    self.__load_mat_file(url)
  File "/home/gromualdi/robot-code/robotology-superbuild/build/install/lib/python3/dist-packages/robot_log_visualizer/ui/gui.py", line 584, in __load_mat_file
    self.signal_provider.open_mat_file(file_name)
  File "/home/gromualdi/robot-code/robotology-superbuild/build/install/lib/python3/dist-packages/robot_log_visualizer/file_reader/signal_provider.py", line 160, in open_mat_file
    self.data = self.__populate_numerical_data(file)
  File "/home/gromualdi/robot-code/robotology-superbuild/build/install/lib/python3/dist-packages/robot_log_visualizer/file_reader/signal_provider.py", line 153, in __populate_numerical_data
    data[key] = self.__populate_numerical_data(file_object=value)
  File "/home/gromualdi/robot-code/robotology-superbuild/build/install/lib/python3/dist-packages/robot_log_visualizer/file_reader/signal_provider.py", line 153, in __populate_numerical_data
    data[key] = self.__populate_numerical_data(file_object=value)
  File "/home/gromualdi/robot-code/robotology-superbuild/build/install/lib/python3/dist-packages/robot_log_visualizer/file_reader/signal_provider.py", line 137, in __populate_numerical_data
    if data[key]["timestamps"][0] < self.initial_time:
IndexError: too many indices for array: array is 0-dimensional, but 1 were indexed

```

closed 2024-12-18 by GiulioRomualdi